lowRISC/sonata-system #15: mdBook copies everything to book folder

marnovandermaas, 2023-12-22:
During the documentation building, mdBook copies everything from the root directory into the `book` directory. I found this out because FuseSoC was using an old core file located in the `book` directory instead of using the most recent changes in root.

Is there any way that we can fix this? Should we limit mdBook generation to just within the `doc` folder?

marnovandermaas, 2023-12-22:
@HU90m, @hcallahan-lowrisc and @GregAC for visibility.

HU90m, 2023-12-23:
This is a known [complaint](https://github.com/rust-lang/mdBook/issues/1187) upstream. I have a nix build that ignores all files that aren't related to documentation. I'll open a PR once I get the preprocessors working.

hcallahan-lowrisc, 2023-12-23:
> This is a known [complaint](https://github.com/rust-lang/mdBook/issues/1187) upstream.

Yeah it would be nice to have an upstream fix for this. It is definitely inconvenient when your docs are interspersed with code. I think fusesoc has a FUSESOC_IGNORE file or something similar to exclude a directory from searching for .core files, that could be a band-aid for now?

> I have a nix build that ignores all files that aren't related to documentation. I'll open a PR once I get the preprocessors working.

I have a branch with a working nix flake to build the OpenTitan site, might be helpful as a reference for the preprocessors. https://github.com/hcallahan-lowrisc/opentitan/blob/nix_site_build/util/site/flake.nix

HU90m, 2024-01-04:
A little update.

I have a nix build that works around this in a branch of symphony: https://github.com/HU90m/symphony-system/tree/nix-mdbook

I'd like to get to move this nix app into lowrisc-nix because it will be common to both sonata and symphony (and possibly future repos as well).

marnovandermaas, 2024-02-23:
I'm going to close this issue since the documentation update https://github.com/lowRISC/sonata-system/pull/19, the issue already being tracked upstream and the potential nix workaround.

closed 2024-02-23 by marnovandermaas