lowRISC/sonata-system #39: pip3 install -r python-requirements.txt fails

janhoogerbrugge, 2024-04-03:
(.venv) ~/sonata-system % pip3 install -r python-requirements.txt
Collecting git+https://github.com/lowRISC/edalize.git@ot (from -r python-requirements.txt (line 13))
  Cloning https://github.com/lowRISC/edalize.git (to revision ot) to /tmp/pip-req-build-zpcmnb8z
  Running command git clone -q https://github.com/lowRISC/edalize.git /tmp/pip-req-build-zpcmnb8z
  Running command git checkout -b ot --track origin/ot
  Switched to a new branch 'ot'
  Branch 'ot' set up to track remote branch 'ot' from 'origin'.
Collecting git+https://github.com/lowRISC/fusesoc.git@ot (from -r python-requirements.txt (line 16))
  Cloning https://github.com/lowRISC/fusesoc.git (to revision ot) to /tmp/pip-req-build-twxj7lb5
  Running command git clone -q https://github.com/lowRISC/fusesoc.git /tmp/pip-req-build-twxj7lb5
  Running command git checkout -b ot --track origin/ot
  Switched to a new branch 'ot'
  Branch 'ot' set up to track remote branch 'ot' from 'origin'.
Collecting anytree
  Using cached anytree-2.12.1-py3-none-any.whl (44 kB)
Collecting hjson
  Using cached hjson-3.1.0-py3-none-any.whl (54 kB)
Collecting mako
  Using cached Mako-1.3.2-py3-none-any.whl (78 kB)
Collecting pyyaml
  Using cached PyYAML-6.0.1-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (736 kB)
Collecting wheel
  Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
Collecting Jinja2>=2.11.3
  Using cached Jinja2-3.1.3-py3-none-any.whl (133 kB)
Collecting pyparsing
  Using cached pyparsing-3.1.2-py3-none-any.whl (103 kB)
Collecting simplesat>=0.8.0
  Using cached simplesat-0.8.2-py2.py3-none-any.whl (246 kB)
Collecting six
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting MarkupSafe>=0.9.2
  Using cached MarkupSafe-2.1.5-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (26 kB)
Collecting attrs>=17.4.0
  Using cached attrs-23.2.0-py3-none-any.whl (60 kB)
Collecting okonomiyaki>=0.16.6
  Using cached okonomiyaki-1.4.0-py2.py3-none-any.whl (3.1 MB)
Collecting jsonschema>=2.5.1
  Using cached jsonschema-4.21.1-py3-none-any.whl (85 kB)
Collecting zipfile2>=0.0.12
  Using cached zipfile2-0.0.12-py2.py3-none-any.whl (44 kB)
Collecting distro; sys_platform == "linux"
  Using cached distro-1.9.0-py3-none-any.whl (20 kB)
Collecting referencing>=0.28.4
  Using cached referencing-0.34.0-py3-none-any.whl (26 kB)
Collecting jsonschema-specifications>=2023.03.6
  Using cached jsonschema_specifications-2023.12.1-py3-none-any.whl (18 kB)
Collecting importlib-resources>=1.4.0; python_version < "3.9"
  Using cached importlib_resources-6.4.0-py3-none-any.whl (38 kB)
Collecting rpds-py>=0.7.1
  Using cached rpds_py-0.18.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.1 MB)
Collecting pkgutil-resolve-name>=1.3.10; python_version < "3.9"
  Using cached pkgutil_resolve_name-1.3.10-py3-none-any.whl (4.7 kB)
Collecting zipp>=3.1.0; python_version < "3.10"
  Using cached zipp-3.18.1-py3-none-any.whl (8.2 kB)
Building wheels for collected packages: edalize, fusesoc
  Building wheel for edalize (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/jan/sonata-system/.venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-zpcmnb8z/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-zpcmnb8z/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-gquf3s7q
       cwd: /tmp/pip-req-build-zpcmnb8z/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for edalize
  Running setup.py clean for edalize
  Building wheel for fusesoc (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/jan/sonata-system/.venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-twxj7lb5/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-twxj7lb5/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-we9a11kf
       cwd: /tmp/pip-req-build-twxj7lb5/
  Complete output (23 lines):
  WARNING: The wheel package is not available.
  WARNING: The wheel package is not available.
  WARNING: The wheel package is not available.
  WARNING: The wheel package is not available.
  /tmp/pip-req-build-twxj7lb5/.eggs/setuptools_scm-8.0.4-py3.8.egg/setuptools_scm/_integration/setuptools.py:30: RuntimeWarning:
  ERROR: setuptools==44.0.0 is used in combination with setuptools_scm>=8.x
  
  Your build configuration is incomplete and previously worked by accident!
  setuptools_scm requires setuptools>=61
  
  Suggested workaround if applicable:
   - migrating from the deprecated setup_requires mechanism to pep517/518
     and using a pyproject.toml to declare build dependencies
     which are reliably pre-installed before running the build tools
  
    warnings.warn(
  WARNING setuptools_scm.pyproject_reading toml section missing 'pyproject.toml does not contain a tool.setuptools_scm section'
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for fusesoc
  Running setup.py clean for fusesoc
Failed to build edalize fusesoc
Installing collected packages: six, anytree, hjson, MarkupSafe, mako, pyyaml, wheel, Jinja2, edalize, pyparsing, attrs, rpds-py, referencing, zipp, importlib-resources, jsonschema-specifications, pkgutil-resolve-name, jsonschema, zipfile2, distro, okonomiyaki, simplesat, fusesoc
    Running setup.py install for edalize ... done
    Running setup.py install for fusesoc ... done
Successfully installed Jinja2-3.1.3 MarkupSafe-2.1.5 anytree-2.12.1 attrs-23.2.0 distro-1.9.0 edalize-0.3.3 fusesoc-0.1 hjson-3.1.0 importlib-resources-6.4.0 jsonschema-4.21.1 jsonschema-specifications-2023.12.1 mako-1.3.2 okonomiyaki-1.4.0 pkgutil-resolve-name-1.3.10 pyparsing-3.1.2 pyyaml-6.0.1 referencing-0.34.0 rpds-py-0.18.0 simplesat-0.8.2 six-1.16.0 wheel-0.43.0 zipfile2-0.0.12 zipp-3.18.1
(.venv) ~/sonata-system % 

I am using Ubuntu 20.04.

GregAC, 2024-04-03:
Thanks for the report @janhoogerbrugge. Please note the sonata system is still in an early development phase and is not ready for wider use. This is reflected not only in functionality of the system itself but also in documentation, build and setup components. We will be providing documentation, getting started guides and all of the build/setup tooling around it but it's not the priority right now. As such you may find multiple issues when trying to do your own build.

marnovandermaas, 2024-04-04:
I've seen this error before. You need to execute the same command again, then it should succeed.

closed 2024-04-04 by marnovandermaas

janhoogerbrugge, 2024-04-04:
Indeed if I execute it once more then it succeeds. 

I did this before reporting this issue but then fusesoc was failing with:

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa8 in position 72: invalid start byte

My first reaction was that this was because of the error reported during installation of the packages. 

When I tried to debug the error I discovered that the error was because fusesoc scans for .core files and found many of them in the cheriot-llvm directory. They have the same extension but the content is very different ;-). After moving the cheriot-llvm directory to another place fusesoc seem to run fine.

marnovandermaas, 2024-04-04:
Yes, this is a downside of fusesoc unfortunately. You can always place a file named `FUSESOC_IGNORE` in folders that you don't want fusesoc to browse.