lowRISC/sonata-system #91: Where is the bitstream?

benlaurie, 2024-05-23:
https://github.com/lowRISC/sonata-system/blob/main/doc/guide/fpga-update.md says

"Download lowrisc_sonata_system_0.bit from THEBESTURL.COM/lowrisc_sonata_system_0.bit"

Needless to say, that URL doesn't work (though apparently I could buy the domain for $995). So, where does one download this?

marnovandermaas, 2024-05-23:
I just made a bitstream release here: https://github.com/lowRISC/sonata-system/releases/tag/v0.1 I'll update the instructions.

colinoflynn, 2024-05-23:
FYI @marnovandermaas - quickly tried the quickstart as well on that release (reprogramming the RP2040 with your example) and it wasn't loading from SPI flash, on serial I just got:
```
[1B][35mbootloader[1B][0m: Loading software from flash...
[1B][35mbootloader[1B][0m: Failed Magic Check
```

marnovandermaas, 2024-05-23:
I opened up a PR with some doc fixes.

marnovandermaas, 2024-05-23:
> FYI @marnovandermaas - quickly tried the quickstart as well on that release (reprogramming the RP2040 with your example) and it wasn't loading from SPI flash, on serial I just got:
> 
> ```
> [1B][35mbootloader[1B][0m: Loading software from flash...
> [1B][35mbootloader[1B][0m: Failed Magic Check
> ```

Did you also drag `sonata_simple_demo_v0.1.uf2` into the SONATA drive? And otherwise try again after replug?

closed 2024-05-24 by marnovandermaas