lowRISC/sonata-system #95: USB mount is unreliable

benlaurie, 2024-05-24:
I am using an M1 Mac. If I boot the board in the RP2040 mode (as described at https://github.com/lowRISC/sonata-system/releases/tag/v0.1), the mount appears reliably every time. However, in the other (Sonata?) mode, the mount only appears about once in 10 attempts (luckily it did work the very first time or I would've given up).

I've also tested this on an M2 Mac and it was also unreliable.

marnovandermaas, 2024-05-24:
Thanks for reporting this. I did a test on my x86 Mac at home and the `SONATA` drive mounts reliably. The only thing I noticed was that it doesn't remount automatically after the `RPI-RP2` drive dismounts. Other than that it pops up on every disconnect and reconnect. We'll need to test this further on an Arm based Mac.

colinoflynn, 2024-05-27:
More details are available in the [sonata-rp2040 ticket ](https://github.com/newaetech/sonata-rp2040/issues/9) now.

colinoflynn, 2024-05-27:
This should work with programming [release 0.2 of the RP2040 ](https://github.com/newaetech/sonata-rp2040/releases/tag/v0.2.0) (step 1, holding down the `RP2040 Boot` button).

For some reason the bitstream programming seems to be unreliable with drag & drop. But seems to work from the terminal:
1. Double-click on SONATA drive on the Desktop to mount
2. `cp ~/Downloads/sonata_bitstream_v0.1.bit /Volumes/SONATA/.`

The UF2 seems to work fine with graphical drag & drop.

This will be investigated but likely will be later before this is fixed. In the meantime hopefully this gets things back working!

nbdd0121, 2024-05-27:
I think this flakiness is fundamentally intrinsic to how the UF2 file is handled. The current code assumes that blocks will be written in order, which isn't true in most cases.

alex-dewar, 2024-05-27:
I don't think that's the reason. Both UF2 programming and bitstream programming assume that blocks are written in order, but UF2 programming works and bitstream programming doesn't. I think it's more likely to have to do with something like the device writing to the file table or file metadata during the programming, as this would be caught by the current logic with UF2 files but not with bitstreams.

nbdd0121, 2024-05-27:
I can't get either bitstream programming or UF2 programming to work on my machine.

Programming UF2 reports success in the log but the content written is bad and bootloader rejects it. Bitstream programming fails with CRC error.

By choosing to expose as mass storage device there's no guarantee that anything written will be in order.

marnovandermaas, 2024-06-03:
There's a workaround for this, please use the temporary flow described in this release: https://github.com/lowRISC/sonata-system/releases/tag/v0.2

marnovandermaas, 2024-10-11:
I'm closing this issue now since this should be fixed since https://github.com/lowRISC/sonata-system/releases/tag/v0.4.1
Feel free to re-open if any new issues appear.

closed 2024-10-11 by marnovandermaas