lowRISC/sonata-system #96: Examples don't build

benlaurie, 2024-05-24:
```
$ xmake config --board=sonata
checking for platform ... cheriot
checking for architecture ... cheriot
generating /Users/benl/git-work/sonata-software/cheriot-rtos/sdk/firmware.ldscript.in ... ok
(.venv) benl-macbookpro6:sonata-software benl$ xmake build
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/crt/cz.c
[ 41%]: cache compiling.release cheriot-rtos/sdk/core/allocator/main.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/core/loader/boot.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/core/software_revoker/revoker.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/compartment_helpers/check_pointer.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/compartment_helpers/claim_fast.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/locks/semaphore.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/atomic/atomic1.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/crt/arith64.c
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/atomic/atomic4.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/lib/locks/locks.cc
[ 41%]: compiling.release cheriot-rtos/sdk/core/loader/boot.S
error: clang: error: unknown argument: '-mxcheri-rvc'
clang: error: unsupported argument 'cheriot' to option '-mcpu='
  > in cheriot-rtos/sdk/lib/crt/cz.c
```

I presume this is because it is not using the CHERIoT LLVM, but I don't see anything in the setup that would have made it available...

benlaurie, 2024-05-24:
LOL! No: you only added that to the software setup 2 hours ago!

marnovandermaas, 2024-05-24:
The manual setup should work if you want to build the baremetal examples located in the `sw` repository in this repo. `xmake` is only needed in the separate `sonata-software` repository. Maybe this needs to be clarified in the documentation.

benlaurie, 2024-05-24:
So, I did the Nix thing, and I get the same error:

```
(nix:sonata-sw-env) benl-macbookpro6:sonata-software benl$ xmake build
[ 25%]: cache compiling.release compartments/echo.cc
[ 26%]: compiling.release cheriot-rtos/sdk/core/token_library/token_unseal.S
[ 28%]: compiling.release cheriot-rtos/sdk/core/switcher/entry.S
[ 29%]: cache compiling.release cheriot-rtos/sdk/core/software_revoker/revoker.cc
[ 31%]: cache compiling.release cheriot-rtos/sdk/lib/crt/cz.c
[ 32%]: cache compiling.release cheriot-rtos/sdk/lib/crt/arith64.c
[ 34%]: cache compiling.release cheriot-rtos/sdk/lib/atomic/atomic1.cc
[ 35%]: cache compiling.release cheriot-rtos/sdk/core/scheduler/main.cc
[ 37%]: cache compiling.release cheriot-rtos/sdk/lib/atomic/atomic4.cc
[ 38%]: cache compiling.release cheriot-rtos/sdk/lib/locks/locks.cc
[ 40%]: cache compiling.release cheriot-rtos/sdk/lib/locks/semaphore.cc
[ 41%]: cache compiling.release cheriot-rtos/sdk/core/allocator/main.cc
error: clang: error: unknown argument: '-mxcheri-rvc'
clang: error: unsupported argument 'cheriot' to option '-mcpu='
clang: error: unsupported argument 'cheriot' to option '-mcpu='
  > in cheriot-rtos/sdk/core/token_library/token_unseal.S
```

marnovandermaas, 2024-05-27:
I'm assuming you are running this in the [software repository](https://github.com/lowRISC/sonata-software). This looks like something went wrong in setting up the Nix environment.
```
$ clang --version
clang version 13.0.0
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /nix/store/m45jn5dx5x063nr76zs8gkg1yrcqrzhc-llvm-cheriot-13.0.0/bin
```

The other thing I can suggest is making sure that you are not defining `CHERIOT_RTOS_SDK` or `CHERIOT_LLVM_BIN` inside your Nix environment, see https://github.com/lowRISC/sonata-system/issues/101.

benlaurie, 2024-05-27:
What do you mean by "*in* the software repository"?

benlaurie, 2024-05-27:
Also:

```
$ clang --version
clang version 13.0.0
Target: aarch64-apple-darwin23.5.0
Thread model: posix
InstalledDir: /nix/store/r60i4zgahhw37q9ljsprkz6b25fhp4ld-llvm-cheriot-13.0.0/bin
```

benlaurie, 2024-05-27:
BTW, I also had to `$ export NIXPKGS_ALLOW_UNSUPPORTED_SYSTEM=1` and `$ nix develop github:lowRISC/sonata-software --impure` or it wouldn't build because it said tbox was not avaikale: `error: Package ‘tbox-1.7.5’ in /nix/store/b2hxgkvh0jrkxfv312xcrmrr3qyz50c1-source/pkgs/development/libraries/tbox/default.nix:30 is not available on the requested hostPlatform`

marnovandermaas, 2024-06-03:
Thanks for reporting those Nix issues. We should now have this fixed.

Since you're using `xmake`, I assume you are using the instructions this repository: https://github.com/lowRISC/sonata-software/blob/main/README.md

We have recently updated those, so can you try running them again?

marnovandermaas, 2024-10-18:
This should now be fixed with the newest getting started guide, feel free to reopen if it isn't.

closed 2024-10-18 by marnovandermaas