lowRISC/sonata-system #97: xmake is not listed as a prerequistie

benlaurie, 2024-05-24:


marnovandermaas, 2024-05-24:
Thanks Ben! I recently split the software toolchain setup page into a Nix and a manual setup. The Nix setup should point you to the sonata software repository and `xmake` is included in the Nix environment: https://lowrisc.org/sonata-system/doc/guide/toolchain-setup.html?highlight=requi#nix-setup

benlaurie, 2024-05-24:
OK, but it isn't included in the manual setup.

davidchisnall, 2024-07-11:
Please can you point people at the CHERIoT docs, rather than trying to duplicate half of them?

marnovandermaas, 2024-10-18:
Ok, I've listed xmake as a dependency for Sonata software and have linked to the CHERIoT RTOS documentation for a full development setup: https://github.com/lowRISC/sonata-system/pull/260

closed 2024-10-21 by marnovandermaas