lowRISC/sonata-system #119: UART output

janhoogerbrugge, 2024-06-05:
If I print to the UART (the one via the usb port that is also powering the board) at full speed then the output becomes garbage. I use GTKTerm. If I switch to RTS/CTS or Xon/Xoff flow control then it sometimes works but also sometimes not? Is this known?

Furthermore, what is needed to make Debug from CHERIoT RTOS working so that I can do Debug::log(...)?

What is not supported from the CHERIoT RTOS?

davidchisnall, 2024-07-11:
> If I switch to RTS/CTS or Xon/Xoff flow control then it sometimes works but also sometimes not?

The UART doesn't do any (hardware or software) flow control, so these should both be disabled in your terminal.  The default speed is 115200 Baud 8N1.

> Furthermore, what is needed to make Debug from CHERIoT RTOS working so that I can do Debug::log(...)?

The `Debug` type is usually an alias for a `ConditionalDebug` template instantiation.  Whether you get output is controlled by the first template parameter.  For most of the examples, this is unconditionally true.  For the core components, this is configured via the various debug flags in the configuration step.

> What is not supported from the CHERIoT RTOS?

I believe all of the RTOS features are supported on Sonata (except the hardware revoker without #141).  We have drivers for the UART and Ethernet adaptor, the D-pad and the LEDs.  We do not (yet) have drivers for the LED screen.  The RTOS test suite all passes.  See: https://cheriot.org/fpga/ibex/2024/06/10/sonata-quick-start.html

marnovandermaas, 2024-10-21:
I am going to close this issue because I think this has been solved. The baud rate has since been increased to 921600 in the v0.4 release, but everything else should be the same. If you have any more issues feel free to reopen this issue or create a new one!

closed 2024-10-21 by marnovandermaas