lowRISC/sonata-system #136: Shouldn't the Ibex be configured as RV32E=1'b1?

ColinSCISemi, 2024-07-03:
Hi,

CHERIoT is spec'ed as being 16 GP registers only. However, in Sonata the RV32E is set to 0 which means that there are 32 GP registers. Shouldn't the RV32E parameter be set when the CHERIoTEn parameter is set?

Cheers,
Colin

HU90m, 2024-07-03:
Yes indeed, thanks for opening this

marnovandermaas, 2024-07-03:
At the moment the CHERI enable bit is configurable at runtime. The hardware essentially turns on RV32E when CHERI is enabled and otherwise is RV32I. RV32E is a parameter and cannot be changed at runtime. If we go down that route, we should re-do the legacy software part of this repo to work on RV32E first.

marnovandermaas, 2024-07-16:
Also thanks to @ColinSCISemi for creating a patch for CHERIoT Ibex to fix Verilator lint warnings: https://github.com/microsoft/cheriot-ibex/issues/39

GregAC, 2024-07-16:
For sonata-system at least, given it's status as an FPGA based evaluation platform I'd prefer to leave `RV32E` as 0 because of @marnovandermaas's comments above. Whilst we haven't got everything in place to make this work well we want to ensure the released sonata system is able to boot in both CHERIoT mode and vanilla RV32 mode, which should be RV32I given RV32E is little used.

Advanced users of Sonata who wish to optimize resource usage are of course free to ditch the vanilla RV32 mode and set RV32E to 1 and build their own FPGA image. We should make sure it is simple to do this and clear in the documentation it's an option.

marnovandermaas, 2024-10-21:
I'm going to close this issue because I think it is resolved. Please feel free to re-open it if you run into any issues.

closed 2024-10-21 by marnovandermaas