lowRISC/sonata-system #169: Pinmux refinements

marnovandermaas, 2024-08-30:
This issue tracks the outstanding tasks for the pin multiplexer.

- [ ] Use padring design so that the pinmux can be instantiated in `sonata_system` instead of `top_sonata`.
- [ ] Describe the pinmux generation and design in documentation.
- [ ] Generate table of register description for pinmux based on TOML file.
- [ ] Test writing different values to selectors.
- [ ] Reduce number of SPI instances.
- [ ] Allow more flexible use of the Rasbperry Pi HAT and Arduino Shield. For example hook up the PWM.
- [ ] Split top generation from pinmux generation.
- [ ] Route the XADC through the pinmux.

elliotb-lowrisc, 2024-10-03:
Perhaps fix these `input` <-> `inout` mismatches too?
```
WARNING: [Synth 8-6104] Input port 'ser0_rx' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:266]
WARNING: [Synth 8-6104] Input port 'ser1_rx' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:268]
WARNING: [Synth 8-6104] Input port 'rs232_rx' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:270]
WARNING: [Synth 8-6104] Input port 'appspi_d1' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:278]
WARNING: [Synth 8-6104] Input port 'ethmac_cipo' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:281]
WARNING: [Synth 8-6104] Input port 'mb3' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:331]
WARNING: [Synth 8-6104] Input port 'mb8' has an internal driver [/home/eb/sonata-system/build/lowrisc_sonata_system_0/src/lowrisc_sonata_system_0/rtl/fpga/top_sonata.sv:336]
```

HU90m, 2024-10-10:
> Perhaps fix these `input` <-> `inout` mismatches too?
>
> ```
> WARNING: [Synth 8-6104] Input port 'ser0_rx' has an internal driver [...]

👍 fixed this in: https://github.com/lowRISC/sonata-system/pull/226

marnovandermaas, 2024-11-12:
Most of this done and the things that aren't done are split into separate issues.

closed 2024-11-12 by marnovandermaas