lowRISC/sonata-system #373: OpenOCD "load_image" also sets the chip running?

nwf, 2024-12-19:
Not *really* understanding what's going on here, but... the most visible problem is that `util/mem_helper.sh` often fails the verification it tries to do and the chip is clearly running while that's happening.  I'm a little surprised not to see a `reset run` command being given to OpenOCD.

Trying to minimize things a bit, I've observed the following:

1. running
   ```
   openocd -f .../sonata-system/util/sonata-openocd-cfg.tcl -c halt -c exit
   ```
   indeed halts the chip and leaves it that way, even after OpenOCD exits, so presumably `exit` does not imply `reset run`.
2. running
   ```
   openocd -f .../sonata-system/util/sonata-openocd-cfg.tcl -c "load_image .../top_compartment 0" -c "sleep 5000" -c exit
   ```
   loads the image in question and sets the chip in motion at the *start* of the 5 second timeout, suggesting that `load_image` does imply `reset run`, though I don't think it's supposed to.
3. running
   ```
   openocd -f .../sonata-system/util/sonata-openocd-cfg.tcl -c "load_image .../top_compartment 0" -c halt -c "sleep 5000" -c exit
   ```
   does not halt the CPU after `load_image`.  Perhaps some part of the system believes it to be already halted and so does nothing, or perhaps the target is claiming to be halted when it is not.

The upshot of this is that if I remove the `verify_image` commands from `util/mem_helper.sh`, I get much better success rate loading in programs, but this is... disconcerting.  Please advise and please let me know what more I can do or test to be useful.

marnovandermaas, 2024-12-19:
Thanks for sharing this. Halting and exiting is not supported in the current debug setup on Sonata. It is essentially a fancy way of writing to the bus. The way that we "halt" the core is by resetting it any time the debug module writes to the bus. The main reason we designed it this way is that we didn't have a CHERIoT-enabled debug ROM available when we were doing bring-up.

marnovandermaas, 2024-12-19:
Just for reference to anyone else stumbling on this issue, the openOCD flow on Sonata is very much an advanced feature and not the main way we designed Sonata to be used. Most people should just use the UF2 method to load new programs on the board. As far as I understand nwf is wanting to use openOCD to avoid writing too many times to flash.

nwf, 2024-12-19:
AFAICT this is due to https://github.com/lowRISC/sonata-system/blob/e7088a6def91a4fff4c1afe7c8085bf24831d56c/rtl/system/sonata_system.sv#L264-L277 : the chip comes out of reset by itself after the debugger accesses memory?

marnovandermaas, 2024-12-19:
Yes, exactly. The core is held in reset for a while after the debugger has written to the bus. The only way openOCD and/or the debug module can keep the core in reset is by keep on writing stuff to the bus. This allows you to do a load_image and after that whole image is written to SRAM, the core comes out of reset and executes the thing you've just written. This is a developer feature for advanced users only.

marnovandermaas, 2025-03-26:
This code is now removed with this PR: https://github.com/lowRISC/sonata-system/pull/387

closed 2025-03-26 by marnovandermaas