lowRISC/sonata-system #378: Mistaken use of variable assignment in ibex_cs_registers.sv

chris-stafford-64, 2025-01-08:
File ibex_cs_registers.sv contains following:
 logic  cheri_exception_code = mcause_q == 6'h1C;

However, cheri_exception_code is a variable (not a net) and is therefore initialised 'X' at time 0ns and remains 'X'.

Probably really wanted continuous assignment:
logic  cheri_exception_code;
assign cheri_exception_code = mcause_q == 6'h1C;

marnovandermaas, 2025-01-15:
I wonder why this worked before in that case. Either way I've taken your suggestion and fixed it here: https://github.com/lowRISC/cheriot-ibex/commit/97de74dd59688cdb74091cebcc213d4b230cf0e6
This'll be pulled in the next time we pull in the latest version of CHERIoT Ibex.

chris-stafford-64, 2025-01-15:
Thanks Marno.
Yes strange that it has previously worked. Problem was highlighted using Xcelium simulator.

marnovandermaas, 2025-03-26:
Closing this as it is fixed in the Sonata Ibex branch.

closed 2025-03-26 by marnovandermaas