lowRISC/sonata-system #381: Misdirected RAM accesses?

nwf, 2025-01-22:
Experimenting with https://github.com/CHERIoT-Platform/cheriot-rtos/pull/425, I have a build of the `01.hello_world` example that works fine in the sonata simulator (as in the devcontainer, build cdeeb9a17869, from today) but fails to get off the ground on the actual board.

Some _extremely tedious_ debugging [^0] has brought me back to very early parts of the RTOS loader, specifically https://github.com/CHERIoT-Platform/cheriot-rtos/blob/52a1ba418013e1dbe55addf7b7fcf66e96bb4347/sdk/core/loader/boot.S#L45-L59.  In particular, `ca1` correctly ends up being a tagged capability to `0x00101d74`, in SRAM, and subsequent bytes, but the word read back by `clw s0, IMAGE_HEADER_LOADER_CODE_START_OFFSET(ca1)` is sourced from `0x40001d74`, in HyperRAM, and that's the beginning of an extremely rapid unintentional disassembly of the edifice that the loader is attempting to build.

It certainly _appears_ as though something about the actual synthesized gateware, as opposed to the simulated gateware, is misdirecting a read intended for SRAM to HyperRAM.  Of note, the _instructions_ being executed are being fetched from HyperRAM (and apparently, from observation, correctly so).

Does that theory hold water?  Would the `01.hello_world` ELF and/or dump files be useful for investigation?

[^0]: In particular, this is early enough that basically nothing is initialized, but the 8 GPIO LEDs sure do work.  So, 8 bits of information per recompile and code reload... let's not say "fast fluxing" of images, but I was happy to have OpenOCD working.

marnovandermaas, 2025-01-22:
This is very weird. The SRAM and hyperRAM mapping should be the same in simulation and FPGA. Just to double check, can you run your example on the simulator matching the commit of your bitstream?

nwf, 2025-01-22:
H-Hey!!

https://github.com/lowRISC/sonata-system/blob/46497b71c01b090502d4f39d201c50c8f53ff1ab/data/xbar_main.hjson#L234

marnovandermaas, 2025-03-26:
This should be fixed by https://github.com/lowRISC/sonata-system/pull/387

closed 2025-03-26 by marnovandermaas