lowRISC/sonata-system #394: GPIO change notification IRQ(s)?

nwf, 2025-03-26:
It'd be nice if we didn't have to devote thread(s) to polling GPIO devices like the joystick.  Lots of MCUs (and probably the OT GPIO IP block) have the ability to dispatch IRQs on GPIO changes, and it'd be nice if Sonata could, too.

elliotb-lowrisc, 2025-05-23:
I'm having a think about this today. Any suggestions on modes (rising/falling/either/low/high), granularity, etc?

I'm currently thinking of breaking the GPIO down into banks of up to 8 pins grouped by header (RPi0-7, RPi8-15, RPi16-23, RPi24-27, Arduino0-7, Arduino8-13, PMOD0, PMOD1, PMODC, Switches0, Switches1, uSD) for pin-change interrupt purposes only. The GPIO could have one interrupt for all banks/pins. Each bank could be masked. Each pin in each bank could be masked (81 bits packed into 3x 32-bit registers). For mode perhaps rising/falling/either/low. Select between debounced and non-debounced input bank-wise either in the top bits of the bank enable register or a separate register. All very much subject to change. I probably should consider how this would work with compartmentalisation too.

elliotb-lowrisc, 2025-05-23:
Change of plan. Discussing it with others, a per-GPIO-instance approach is probably better. There's currently a different instance per header, although they are all collected together in a single block. There's one for the RPi header, one for Arduino header, one each for PMOD0/1/C, and one for the fixed I/O (LEDs, switches, etc.). Each instance could have an interrupt line, a register masking the input pins, and a control register to set the interrupt mode (rising/falling/etc.). This would probably be a better match to the existing scheme.