lowRISC/sonata-system #401: PLIC autogen RTL is not the result of the current PLIC RTL template and its configuration

nwf, 2025-04-16:
The RISC-V PLIC spec is quite flexible about its [per-source interrupt priorities](https://github.com/riscv/riscv-plic-spec/blob/master/riscv-plic.adoc#interrupt-priorities).  In Sonata, the instantiated widths are uniform across sources, and the choice of that width is nominally managed by the templating machinery, at least as of 423aebb01e261e5f65670c0e70cea4900c9ab867, which predates the 1.0 release.  In particular, it is the `prio` template parameter:

https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/vendor/lowrisc_ip/ip_templates/rv_plic/data/rv_plic.hjson.tpl#L11

Which gets used in a variety of places, including...

* The computation of `PrioWidth`, which then gets fed through other generators (like `reggen`, apparently).

  https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/vendor/lowrisc_ip/ip_templates/rv_plic/data/rv_plic.hjson.tpl#L57-L61

* The width of the PLIC internal registers, presumably also fodder for later stages of the pipeline:

  https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/vendor/lowrisc_ip/ip_templates/rv_plic/data/rv_plic.hjson.tpl#L120-L127

* Directly in the RTL templates:

  https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/vendor/lowrisc_ip/ip_templates/rv_plic/rtl/rv_plic.sv.tpl#L47-L48

Since that commit, the value of that parameter has always been 7:

https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/vendor/lowrisc_ip/ip_templates/rv_plic/data/rv_plic.tpldesc.hjson#L24-L29

and so we should have 3-bit registers all over the place.  However, the checked in generated files are inconsistent with their purported sources:

https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/rtl/system/autogen/rv_plic/rtl/rv_plic_reg_pkg.sv#L12

https://github.com/lowRISC/sonata-system/blob/34c893b1d13b0c5dbacc46a1865e836fca6c05ef/rtl/system/autogen/rv_plic/rtl/rv_plic.sv#L47

And indeed, in practice, 1.0 (and presumably 1.1) bitfiles offer only a 2-bit priority field. :(

marnovandermaas, 2025-04-22:
While investigating this issue I opened this PR: https://github.com/lowRISC/sonata-system/pull/404

nwf, 2025-04-22:
Ah, somehow I missed `rv_plic_cfg.hjson`, perhaps because it was added in a later commit.  Sorry for the noise!

closed 2025-04-22 by nwf