lowRISC/sonata-system #410: The very first Ethernet frame after reset never leaves the device?

nwf, 2025-05-05:
While continuing to investigate network startup time (and https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/issues/1244 in particular), I found what looks like it might be a hardware or driver bug that's beyond my present understanding.  _Specifically, it looks like the Ethernet block fails to send the very first packet it's given._  I am unsure if this is an IP block issue or a driver issue (perhaps we're failing to wait long enough for autonegotiation or link training or somesuch?).  This appears to happen on both warm (`mem_helper.sh`) and cold (power cycle and `mem_helper.sh`) resets of the Sonata.

Turning several debugging knobs up to 11 and adding a few more informative `printf`s, the system spews forth:

* Initialization:
  ```
  Ethernet driver: Acknowledged interrupt, sleeping on futex 0x10bf04 (v:1 0x10bf04-0x10bf08 l:0x4 o:0x0 p: G R----- -- ---)
  vDHCPProcessEndPoint: enter 0
  DHCP-socket[fe-7a]: DHCP Socket Create
  prvCreateDHCPSocket[fe-7a]: open, user count 1
  prvInitialiseDHCP: start after 25 ticks
  vDHCP_RATimerReload: 25
  vDHCPProcessEndPoint: exit 1
  ```

* Attempt to send a DHCP Discover packet without the broadcast-reply flag set (`xUseBroadcast=0`):
  ```
  vDHCPProcessEndPoint: enter 1
  nwf prvSendDHCPDiscover from xHandleWaitingFirstDiscover
  Buffer management: Allocated 0x138 byte buffer: 0x10e630 (v:1 0x10e630-0x10e778 l:0x148 o:0x0 p: G RWcgm- -- ---), descriptor: 0x10e5c8 (v:1 0x10e5c8-0x10e628 l:0x60 o:0x0 p: G RWcgm- -- ---)
  nwf xUseBroadcast=0
  vDHCPProcess: discover
  nwf sendto result 258
  vDHCPProcessEndPoint: exit 2
  Firewall: Permitting broadcast UDP packet
  Firewall: Permitting outbound IPv4 packet
  Ethernet driver: Sending frame of length 0x12c
  Buffer management: Freeing descriptor: 0x10e5c8 (v:1 0x10e5c8-0x10e628 l:0x60 o:0x0 p: G RWcgm- -- ---) and 
  buffer 0x10e630 (v:1 0x10e630-0x10e778 l:0x148 o:0x0 p: G RWcgm- -- ---)
  Ethernet Adaptor: nwf output_frame returning true
  ```

* Timeout, now send a DHCP Discover packet with the broadcast-reply flag set:
  ```
  vDHCPProcessEndPoint: enter 2
  nwf prvSendDHCPDiscover from xHandleWaitingOffer
  Buffer management: Allocated 0x138 byte buffer: 0x10e7e8 (v:1 0x10e7e8-0x10e930 l:0x148 o:0x0 p: G RWcgm- -- ---), descriptor: 0x10e780 (v:1 0x10e780-0x10e7e0 l:0x60 o:0x0 p: G RWcgm- -- ---)
  nwf xUseBroadcast=1
  vDHCPProcess: discover
  nwf sendto result 258
  vDHCPProcess: timeout 1000 ticks
  Firewall: Permitting broadcast UDP packet
  Firewall: Permitting outbound IPv4 packet
  Ethernet driver: Sending frame of length 0x12c
  Buffer management: Freeing descriptor: 0x10e780 (v:1 0x10e780-0x10e7e0 l:0x60 o:0x0 p: G RWcgm- -- ---) and buffer 0x10e7e8 (v:1 0x10e7e8-0x10e930 l:0x148 o:0x0 p: G RWcgm- -- ---)
  Ethernet Adaptor: nwf output_frame returning true
  ```

* _That_ one made it out, and the system gets a reply to it:
  ```
  Ethernet driver: Receiving frame of length 0x15a
  Firewall: DNS server address set to 0x1071dac
  Buffer management: Allocated 0x160 byte buffer: 0x10ea48 (v:1 0x10ea48-0x10ebb8 l:0x170 o:0x0 p: G RWcgm- -- ---), descriptor: 0x10e9e0 (v:1 0x10e9e0-0x10ea40 l:0x60 o:0x0 p: G RWcgm- -- ---)
  Ethernet Adaptor: Sending frame to IP task
  ```
  and so on.

`tcpdump` (no filters beyond the interface, which is set promiscuous) agrees that the first packet it sees from Sonata is a DHCP request _with the broadcast flag set_.  Prior to that, we see the neighbor host doing some MLDv2 announcements as the port comes up.  `3a:30:25:24:fe:7a` is the Sonata; `0c:c4:7a:ab:33:9a` is Sonata's neighbor (bridging Sonata to the larger network); and `b4:fb:e4:20:ca:6b` is the network gateway and DHCP server.

```
16:17:06.755528 0c:c4:7a:ab:33:9a > 33:33:00:00:00:16, ethertype IPv6 (0x86dd), length 90: (hlim 1, next-header Options (0) payload length: 36) fe80::ec4:7aff:feab:339a > ff02::16: HBH (rtalert: 0x0000) (padn) [icmp6 sum ok] ICMP6, multicast listener report v2, 1 group record(s) [gaddr ff02::1:ffab:339a to_ex { }]
16:17:07.421322 0c:c4:7a:ab:33:9a > 33:33:00:00:00:16, ethertype IPv6 (0x86dd), length 90: (hlim 1, next-header Options (0) payload length: 36) fe80::ec4:7aff:feab:339a > ff02::16: HBH (rtalert: 0x0000) (padn) [icmp6 sum ok] ICMP6, multicast listener report v2, 1 group record(s) [gaddr ff02::1:ffab:339a to_ex { }]
16:17:10.407859 3a:30:25:24:fe:7a > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 300: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 286)
    0.0.0.0.68 > 255.255.255.255.67: [no cksum] BOOTP/DHCP, Request from 3a:30:25:24:fe:7a, length 258, xid 0xcae188c, Flags [Broadcast] (0x8000)
          Client-Ethernet-Address 3a:30:25:24:fe:7a
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message (53), length 1: Discover
            Client-ID (61), length 7: ether 3a:30:25:24:fe:7a
            Parameter-Request (55), length 3: 
              Subnet-Mask (1), Default-Gateway (3), Domain-Name-Server (6)
16:17:10.410167 b4:fb:e4:20:ca:6b > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 342: (tos 0xc0, ttl 64, id 16583, offset 0, flags [none], proto UDP (17), length 328)
    172.29.7.1.67 > 255.255.255.255.68: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0xcae188c, Flags [Broadcast] (0x8000)
          Your-IP 172.29.7.97
          Server-IP 172.29.7.1
          Client-Ethernet-Address 3a:30:25:24:fe:7a
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message (53), length 1: Offer
            Server-ID (54), length 4: 172.29.7.1
            Lease-Time (51), length 4: 86400
            RN (58), length 4: 43200
            RB (59), length 4: 75600
            Subnet-Mask (1), length 4: 255.255.255.0
            BR (28), length 4: 172.29.7.255
            Default-Gateway (3), length 4: 172.29.7.1
            Domain-Name-Server (6), length 4: 172.29.7.1
```

marnovandermaas, 2025-05-05:
On Sonata we connect to an ethernet chip (KSZ8851SNLI-TR) via SPI. We control the SPI host RTL and the Ethernet driver that uses it but we do have to deal with a third-party chip here.

marnovandermaas, 2025-05-05:
Also, here's the link to the datasheet, which we should probably put on the documentation somewhere: https://ww1.microchip.com/downloads/aemDocuments/documents/UNG/ProductDocuments/DataSheets/KSZ8851SNL-Single-Port-Ethernet-Controller-with-SPI-DS00002381C.pdf

closed 2025-05-08 by nwf

marnovandermaas, 2025-05-08:
Thanks @nwf for working on this!