lowRISC/sonata-system #429: Enable 8-bit and 16-bit memory accesses via the SBA interface to the debug module

resistor, 2025-06-25:
Today the `sbaccess16` and `sbaccess8` fields in the debug module's `sbcs` register are fixed to 0: https://github.com/lowRISC/sonata-system/blob/5683774c60883e7a98883a5a8d98e2c75da09eee/vendor/cheriot_debug_module/src/dm_csrs.sv#L525

This prevents memory reads and writes of smaller that 32 bits from being completed using the SBA interface, and particularly blocks the ability to implement software breakpoints, as that requires reading/writing 16 bit instructions. While hardware supported breakpoints are available, they are limited in quantity, so we would like to verify that software breakpoints work as an alternative.

Is it possible to enable these smaller SBA access sizes?

elliotb-lowrisc, 2025-06-25:
Thanks for reporting your issue.

I see that `sbaccess16` is assigned `0` in the CHERIoT-Platform debug module we're using (https://github.com/CHERIoT-Platform/cheriot-dbg-module/blob/845f74ce6deb8e811f695c8dab1e6b4ba9cf2fbd/src/dm_csrs.sv#L525), but not in the upstream PULP one (https://github.com/pulp-platform/riscv-dbg/blob/618ee6e0e2610ef47e0dcc4df6748af3dffff731/src/dm_csrs.sv#L566) since 2021 (https://github.com/pulp-platform/riscv-dbg/commit/ad0fbea2467db66cad90c1689974962abeeffe1d). The first question I have is why the difference exists.

Do you know (seeing as you are at SCI, I presume) if it was simply that the CHERIoT-Platform debug module is based on an older version of the PULP one, or is there a technical reason for it?

In either case, I expect more investigation and testing will be required.

elliotb-lowrisc, 2025-06-26:
Thanks, good to know. Hopefully there's no reason this can't work then.

I suppose the next step is updating the CHERIoT-Platform debug module to support 16-bit and 8-bit accesses, followed by pulling the updated module into sonata-system and fixing any integration issues. A test case or other instructions would be useful for the latter, if you have something we could use.

As for the former (updating the debug module), is that something you can do seeing as you have access to CHERIoT-Platform repos? If not, perhaps Colin could as the one who ported it. I may be able to help, but my knowledge of the debug module is probably lacking.

elliotb-lowrisc, 2025-09-04:
Any update on getting this change into the debug module?

resistor, 2025-09-04:
I spoke to Colin about chatting with you all about this yesterday, so hopefully he'll be able to weigh in soon.

resistor, 2025-09-04:
The symptoms are distinct, but they're both downstream of the debug module in Sonata and Iceni having diverged. As such I expect that solving that root problem will let us address both.